sbgisen/vesc #5: Change the algorithm for homing operation.

nyxrobotics, 2020-05-14:
### 現在の原点復帰動作
- 動き出し始める最小限の電流で電流制御を行い、弱い力で機械的に原点位置にぶつけにいく
- ホールセンサのカウント値が変化しなくなったことを検知して原点位置にぶつかったことを判定して終了する

この原点復帰手順だと以下の課題が発生したため、動作原理を変更する必要が有る
- 原点復帰動作に失敗した際にモータが燃える（原因は不明。おそらく原点位置にぶつかった際のノイズか何かで通信エラーしたものと思われる）
  - VESC本体の設定でタイムアウト処理ができるのが理想だが、これに関する設定はvesc_tool上では見つからなかった
  - モータにサーミスタをつける（VESCのTEMP_MOTORピンを使用する）
  - 原点復帰動作時のみ、電流制限を厳しくする（効果が有るか不明）
- コンディションによって「大きく加速してしまう」/「全く動かない」という状況が発生する
  - 電流のみを制御しているため、摩擦や粘性の状態に大きく依存している
  - 特にモータの回転方向を間違って指定した際、自重が加わって逆方向に加速してしまうため危険な動作をする

nyxrobotics, 2020-05-14:
## 案１
1. 最大電流を小さくする
1. Duty比を指定して原点復帰動作を行う（これだと速度フィードバックをしていないので根本的には解決できていない。電圧は最大回転数と比例するので、電流司令時の無制限な加速はなくなる）
1. Duty比が固定の状態で、電流が一定より大きくなったことを検知して原点にぶつかったことを判別する
1. 原点復帰動作を終了し、最大電流を元に戻す

nyxrobotics, 2020-05-14:
原点復帰位置より少し内側の位置を初期目標位置・可動範囲の終端とすることでモータが燃える問題は解決できそうです

ssr-yuki, 2021-05-27:
Duty-based calibration had been implemented by #15.

closed 2022-10-19 by nyxrobotics