sbgisen/vesc #10: VESC Not working fully with jetson TX2

areebsh1, 2020-08-25:
Hi there,
I m building RC car considering the all the steps in F1tenth. Furthermore, I used your vesc package. Actually, I m facing the following problem and I will be great to help me

When I m connecting VESC directly to my laptop( also ubuntu 18.04) and running VESC driver node and publishing speed parameter as:  rostopic pub -r 10 /commands/motor/speed std_msgs/Float64 "data: -2000.0"
, it looks every thing fine and the car driving backwards. Also, it drives forwards in case of publishing 2000.0 instead of -2000.0, no problem with that.

However, When I m connecting VESC directly to Jetson and running VESC driver node and publishing speed parameter as:  rostopic pub -r 10 /commands/motor/speed std_msgs/Float64 "data: -2000.0", the car does not drive backwards. But it drives forwards in case of publishing 2000.0 instead of -2000.0

Any suggestion about why I can turn the motor in the backwards direction in the situation VESC connected to laptop, while I fail to do so when VESC connecting to Jeston? why Jetson does not react to -2000.0 while it reacts to 2000.0?

I m using the following tools:
1. Jetson TX2 flashed with jetpack 4.4 (ubuntu 18.04).
2. VESC MKIV 
3. Traxxas Velineon motor

ssr-yuki, 2020-10-16:
Thank you for reporting. Sorry, I have no idea to solve the problem immediately.

If you have time, please try following:
- assign other negative values (e.g. -100,  -10000) and check screen output messages
- check whether topics surely reach vesc_driver node

If the factor of the problem lies on this packages, some undefined behaviours cause it just on Jetson architecture, I think. More information is needed.

SMemory, 2021-02-06:
Hi, the following method may help.
in file vesc_driver/src/vesc_packet.cpp line 413
change
uint32_t v = static_cast<uint32_t>(getRpm);
to
int32_t v = static_cast<int32_t>(getRpm);

closed 2021-06-07 by ssr-yuki