sbgisen/vesc #18: Pole pairs in eprm calculations

JuliusSustarevas, 2021-05-10:
Hi, 
I might be miles off here, but I think the getRpm()  and setRpm functions should also take into account pole pairs.     Is this done somewhere else? My empirical test using the velocity interface shows a discrepancy that seems to match the pole pair count.  Am I wrong here

ssr-yuki, 2021-06-07:
VESC returns velocity information in RPM unit. Thus the discrepancy is not a bug of drivers but one of VESC firmware or its setting, I think.

However, setting motor poles with vesc_tool does not work in my environments. Could you tell me how to measure real RPM in your test?

ssr-yuki, 2021-06-21:
I completely understand what you say: VESC returns velocity status in ERPM, i.e. mechanical motor velocity [RPM] * the number of motor pole pairs. 

Now I start to work on this issue.

JuliusSustarevas, 2021-06-21:
Sorry, for no reply!  But yes, thats what I meant, I ended up just adding
this as a param on interface, but vesc must know the pole pairs and maybe
it can be read from the message.

closed 2021-06-24 by nyxrobotics

ssr-yuki, 2021-06-24:
@JuliusSustarevas We have corrected codes for this issue.
Please check if you still have interest. Thank you for your report.

PR #21 (2021-06-24): Correct velocity units with the # of motor pole pairs
closes #18  and #20 

- corrects velocity units by parameterization of the number of motor pole pairs
- renames variables and functions with `RPM` to nemes with `ERPM`
- enables ROS interfaces to use SI units, i.e. [rad/s]

For sample execution, please execute the following command:
```bash
roslaunch vesc_driver vesc_driver_node.launch
```