sbgisen/vesc #33: ERROR: cannot launch node of type [vesc_hw_interface/vesc_hw_interface_node]

matthewryanburgin191, 2021-07-25:
Good day,

When roslaunch vesc_hw_interface velocity_control_sample.launch or other examples. the following error occurs;

ERROR: cannot launch node of type [vesc_hw_interface/vesc_hw_interface_node]: Cannot locate node of type [vesc_hw_interface_node] in package [vesc_hw_interface]. Make sure file exists in package path and permission is set to executable (chmod +x)

The vesc_hw_interface_node.cpp has executable permissions.

what am i missing?

best regards
matthew

ssr-yuki, 2021-07-29:
Hi, thank you for reporting. 

Unfortunately, I cannot reproduce such a error on my several computers. Therefore I would like to ask you the following two questions:

1. Have you run `source {YOUR_CATKIN_WS_PATH}/devel/setup.bash` surely after building?
2. Does a binary named `vesc_hw_interface_node`  exist in `{YOUR_CATKIN_WS_PATH}/devel/lib/vesc_hw_interface`?

matthewryanburgin191, 2021-08-01:
Thank you for your replay.
1. Yes its sourced.
2. No there is no binary of that name.

I've tried it on 2 Pc's amd64 running Ubuntu and melodic with same results and with multiple rebuilds.

How has the binary been missed? Is it a catkin issue?

Best regards 

Matthew

matthewryanburgin191, 2021-08-02:
Sorry my appologies!
/home/pi/catkin_ws/devel/lib/vesc_hw_interface/vesc_hw_interface_node
is present

matthewryanburgin191, 2021-08-02:
Solved-ish:
So I went and built another package in the catkin work space and when launched it had a very simular error.
After deleting the workspace for the second or third time the error disappeared and the node launches.

Best regards

ssr-yuki, 2021-08-03:
That's a relief. This issue must come from not this repo. but ROS path system.

closed 2021-08-03 by ssr-yuki