sbgisen/vesc #40: Reverse Control

Parker-Bergen, 2021-11-06:
## Abstract
Reverse motor spin while using current control with the vesc_driver node

## Factors and Question
I am trying to provide a negative current to go reverse, however I'm not sure this is correct and it doesn't work.
Am I missing something like limits or is there a way to tell the controller to reverse the direction?

ssr-yuki, 2021-11-08:
> I am trying to provide a negative current to go reverse, however I'm not sure this is correct ...

You are right. Either your VESC board or the ROS driver in this package does not work.

Can you drive your motor with any other tool such as vesc_tool with both positive and negative current commands?

Parker-Bergen, 2021-11-14:
Okay sorry for the late response. I have tested negative current with the VESC tool, this does make the motor go in the reverse direction, therefore leading to the VESC board to be okay and a valid command.

ssr-yuki, 2021-11-18:
Thank you for reporting. The reported bug hears the same one in #10 .
(I closed that issue since the bug reporter never replies.)

Yes, I guess the bug would hide in vesc_driver.

Could you try to replace `uint32` with `int32` in the following line?
https://github.com/sbgisen/vesc/blob/00424815f1fefc120020434a617d1edc41020f72/vesc_driver/src/vesc_packet.cpp#L381

Parker-Bergen, 2021-11-20:
Changing that line to an int did solve the issue and the motor can now rotate in reverse.

ssr-yuki, 2021-11-24:
Thank you for checking so much, and I am glad to hear that.
I will commit to fix the problem.

ssr-yuki, 2021-12-06:
@Parker-Bergen 
Could you tell me the architecture type of CPU you use? (E.g. ARM, Intel, etc.)

ssr-yuki, 2021-12-09:
Note that this issue was solved by #41

Parker-Bergen, 2021-12-21:
The CPU I am currently using is ARM

ssr-yuki, 2021-12-31:
@Parker-Bergen Thank you so much for providing information. 
I am sure that this issue comes from implementation-defined behaviors in C++.

closed 2021-12-31 by ssr-yuki