sbgisen/vesc #42: Position unit is revolution not radians

AbdElRahmanFarhan, 2021-12-29:
I think that the driver returns position as a revolution, not radians. I checked the code and I didn't find any conversion from revolution to radians.

ssr-yuki, 2021-12-29:
It is not a bug.

Currently we use vesc_driver as a library, and binaries in vesc_driver package is only for debugging.

See https://github.com/sbgisen/vesc/tree/develop/vesc_hw_interface ; this package supports the unit conversion for various types of hardware integration.

 Or, please contribute if you have problems on that.

AbdElRahmanFarhan, 2021-12-30:
Thanks for pointing that out. So, you mean that the position is actually returned as revolution and I only need to convert it to radians, right?

ssr-yuki, 2021-12-31:
No, in my understanding.
I think that the unit of position is not the number of rotations but the number of hall sensor pulses.

The following codes would help you, where the number of pole pairs and the gear ratio are parameters you have to set.
https://github.com/sbgisen/vesc/blob/3d25c6a61d7cd14c673efe7412cbccd8afb7cad5/vesc_hw_interface/src/vesc_hw_interface.cpp#L218-L225

PR #58 (2022-12-27): fix servo sensing calculation

## Change Summary
* Probably resolve #42 
* resolve #53 
* resolve #56 
* resolve #59 
* Fix the unit conversion from sensor data and adjust the units for each joint type.
* Replace the static variable with members to avoid the thread-sharing bug

## Details

## Impacts

* Probably need to set the proper parameters and tune PID gain again.

## References

## Additional Information

* Note that this only fixes the sensor calculation and, to an extent, the calibration process.
* This is tested with PRISMATIC joints since we use this often with vesc. 
* REVOLUTION j [... cut]
closed 2022-12-27 by nyxrobotics