sbgisen/vesc #43: Firmware 5.03 Support

David9696, 2022-02-28:
Hi,

Thanks for sharing this great project.

Previously, I have been using this package together with my vesc with firmware of 5.02 without issue.

Once I upgraded to 5.03, the motor motion is not smooth as compared to 5.02. Once cmd command is sent the output motion is fast-slow fast-slow. Didn't see similar motion when I was using Vesc app. All the settings are the same for firmware 5.02 and 5.03.

Could it be the vesc firmware issue? Or need to tweak this package to support firmware 5.03?

Thanks.

David

nyxrobotics, 2022-10-07:
First, we have reviewed the packet contents.

https://github.com/sbgisen/vesc/blob/3d25c6a61d7cd14c673efe7412cbccd8afb7cad5/vesc_driver/include/vesc_driver/data_map.h#L23-L63

https://github.com/vedderb/bldc/blob/0b5c9fd0ddefc89ae70ce792baeed303a1f8691b/datatypes.h#L922-L1067

Changed : COMM_SET_ERPM -> COMM_SET_RPM

Other than that, they appear to have just been added.
If anyone can tell me where else it needs to be changed, please reply.

nyxrobotics, 2022-10-07:
What version of vesc tool do you have set up?