sbgisen/vesc #53: rpm calculation is not correct

Tacha-S, 2022-12-09:
## Abstract

Motor does not rotate at correct speed even though `gear_ratio` and `num_motor_pole_pairs` are set correctly.

## How to Reproduce the Bug

## Problem due to the Bug

## Factors

## Suggestion

nyxrobotics, 2022-12-09:
- If the input target value is [rad/sec], it should be divided by the gear ratio here. Also, in other calculations, the pulse counts are calculated by multiplying num_motor_pole_pairs by 3.
  - https://github.com/sbgisen/vesc/blob/84100fae35d2c043a783408363d009ad682171c3/vesc_hw_interface/src/vesc_wheel_controller.cpp#L58
  - https://github.com/sbgisen/vesc/blob/84100fae35d2c043a783408363d009ad682171c3/vesc_hw_interface/src/vesc_wheel_controller.cpp#L72
  - https://github.com/sbgisen/vesc/blob/84100fae35d2c043a783408363d009ad682171c3/vesc_hw_interface/src/vesc_wheel_controller.cpp#L96

nyxrobotics, 2022-12-09:
I haven't tried it, but maybe something like this will fix it.
- ```motor_hall_ppr = 3 * num_motor_pole_pairs_;```
  - I want to try it in a motor with a known number of poles, because I feel it is right to make it 6 instead of 3 here.
- ```target_pulse_ += target_velocity * motor_hall_ppr / (2 * M_PI) / control_rate_ / gear_ratio_;```
- ```velocity_sens_ += this->counterTD(current_pulse, false) * 2.0 * M_PI / motor_hall_ppr * gear_ratio_ * control_rate_;```

nyxrobotics, 2022-12-09:
- The code below is wrong in the unit of radian.
https://github.com/sbgisen/vesc/blob/84100fae35d2c043a783408363d009ad682171c3/vesc_hw_interface/src/vesc_servo_controller.cpp#L300-L301

nyxrobotics, 2022-12-09:
- Reference about nom_motor_pole_pairs
- https://toshiba.semicon-storage.com/jp/semiconductor/knowledge/e-learning/brushless-motor/chapter2/the-relationship-between-poles-phases-and-slots.html

jsupratman13, 2022-12-12:
Ok, so vesc_hw_interface assumes that you are using FOC and hall sensors. Here, vesc_interface's getPosition function returns vesc's tachometer, which is the number of steps the motor has rotated (different from the usual tachometer definition?) as described in 
https://github.com/vedderb/bldc/blob/50e1aee9c929e0e43afbd332e5a9a29df933e795/motor/mcpwm_foc.c#L1298-L1307

Reading the function description (and personally testing), the relation between vesc's tachometer and revolution is probably
```
revolution = vesc tachometer / (number of hall sensors * number of rotor poles)
```
where the `number of hall sensors` is usually 3 and `number of rotor poles` is the number of poles, not pole pairs. Therefore, we can convert revolution to radians (including gear_ratio) with the following:
```
rad = revolution * 2 * pi * gear_ratio
```

nyxrobotics, 2022-12-14:
```motor_hall_ppr = 3 * num_rotor_poles``` or```motor_hall_ppr = 6 * num_rotor_pole_pairs```

PR #55 (2022-12-14): fix rpm calculation

## Change Summary
- partially fix #53
- fix rpm calculation on velocity duty mode

## Details

## Impacts

## References

## Additional Information

- will fix calculation on servo-side on next PR
PR #58 (2022-12-27): fix servo sensing calculation

## Change Summary
* Probably resolve #42 
* resolve #53 
* resolve #56 
* resolve #59 
* Fix the unit conversion from sensor data and adjust the units for each joint type.
* Replace the static variable with members to avoid the thread-sharing bug

## Details

## Impacts

* Probably need to set the proper parameters and tune PID gain again.

## References

## Additional Information

* Note that this only fixes the sensor calculation and, to an extent, the calibration process.
* This is tested with PRISMATIC joints since we use this often with vesc. 
* REVOLUTION j [... cut]
closed 2022-12-14 by nyxrobotics