sbgisen/vesc #67: Fix position_control_sample.launch

nyxrobotics, 2023-01-18:
## Abstract
Specify PID gain in position_control_sample.launch.

## How to Reproduce the Bug

## Problem due to the Bug

## Factors

## Suggestion

jsupratman13, 2023-01-19:
Can you elaborate on this a little bit more?

nyxrobotics, 2023-01-23:
The sample velocity_duty_control_sample.launch has PID gains but position_control_sample.launch doesn't.
Since these are PID-controlled samples, gain adjustment is a must.
It is better to specify gain parameters for position_control_sample.launch.

https://github.com/sbgisen/vesc/blob/3d5e62e736d2c3f92d26e006b3ccf48f5f5c4dc4/vesc_hw_interface/launch/velocity_duty_control_sample.launch#L8-L25

https://github.com/sbgisen/vesc/blob/3d5e62e736d2c3f92d26e006b3ccf48f5f5c4dc4/vesc_hw_interface/launch/position_control_sample.launch#L8-L19

jsupratman13, 2023-01-23:
Fair enough.
Do you have any recommended values or should we use the default values?

PR #71 (2023-01-20): Feature/fix enforce limits

## Change Summary
Fixed a bug in enforceLimits.
Changed the default value of control cycle.

- Fix #67 
- Fix #68 
- Reference: #69 

## Details

## Impacts

## References

## Additional Information
PR #72 (2023-01-30): Feature/smooth diff

## Change Summary
Fixed a bug in the calculation of the differentiation of PID control.
See the issue for details.
~~This PR only fixes vesc_wheel_controller.cpp.~~
~~I need another PR to fix vesc_servo_controller.cpp.~~
Refactored PID controllers for both position control and velocity control.

- Fix #65 
- Fix #67  

## Details
- I have created vesc_step_difference.cpp.
  - Classified functions for differentiating counter values.
- Refactored PID controller.
  - There was a bug in the I-control anti-windup.
  - Unnecessary variables were removed.
- Fixed parameters in la [... cut]
closed 2023-01-30 by jsupratman13