sbgisen/vesc #68: [vesc_hw_interface]enforceLimits() using fixed control period(100Hz)

nyxrobotics, 2023-01-18:
## Abstract
I'm not sure if there is any change in functionality, but looking at the source code, it is odd, so I'm making an issue.

https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_hw_interface.cpp#L244
https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_hw_interface.cpp#L299-L302

Control rate is specified as a parameter in vesc_servo_controller and vesc_wheel_controller
https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_servo_controller.cpp#L55-L56

https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_wheel_controller.cpp#L39

## How to Reproduce the Bug

## Problem due to the Bug

## Factors

## Suggestion

nyxrobotics, 2023-01-18:
Investigating general implementation.
I will try using the argument "period" in write() function.

https://github.com/ros-controls/ros_control/wiki/joint_limits_interface

jsupratman13, 2023-01-19:
To elaborate, the `enforceLimits` control period and the hardware interface node control period are fixed at 100Hz, as shown below:
https://github.com/sbgisen/vesc/blob/1e6ed15aabc3756af8b5e692b08b2e44cb6fdcb8/vesc_hw_interface/src/vesc_hw_interface_node.cpp#L29-L40
https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_hw_interface.cpp#L244
https://github.com/sbgisen/vesc/blob/d96dcc2d81419f33647b37b249769c0101b7b7f5/vesc_hw_interface/src/vesc_hw_interface.cpp#L299-L302

The above should be ok, however, the sample implementation, given in https://github.com/ros-controls/ros_control/wiki/joint_limits_interface, shows the `enforceLimits` taking control period passed from the `write` function. In addition, the [combined hardware interface](https://github.com/ros-controls/ros_control/wiki/Writing-CombinedRobotHW) sample file also shows the control period passed on to the write function. 

Therefore, instead of defining the ros_control's control period on vesc hw interface, better to use the control period given in hardware_interface_node layer.

PR #71 (2023-01-20): Feature/fix enforce limits

## Change Summary
Fixed a bug in enforceLimits.
Changed the default value of control cycle.

- Fix #67 
- Fix #68 
- Reference: #69 

## Details

## Impacts

## References

## Additional Information
closed 2023-01-20 by jsupratman13