sbgisen/vesc #75: Publishing vesc_msgs::VescStateStamped for vesc_hw_interface node

David9696, 2023-02-10:
Hi,

Thanks for sharing your package.

Is it possible to subscribe the vesc_msgs::VescStateStamped(sensors/core topic) when using the vesc_hw_interface node?

Thanks.

jsupratman13, 2023-02-10:
We haven't implemented that feature, but I believe it is possible. May I assume that you want to know vesc's status while using hardware interface?

David9696, 2023-02-10:
Yes, it is good to know the status of vesc while using hardware interface. 

Thanks.

jsupratman13, 2023-02-13:
I have no problem with having this feature. Our hands are tied at the moment, so it would be great if you could set up a Pull Request.

David9696, 2023-02-13:
Hi,

Noted. I am not very good in CPP. But I would try my best to work it out. 

Thanks.

closed 2023-02-13 by David9696