sbgisen/vesc #77: Computed velocity and real velocity not matching in velocity continuous mode

David9696, 2023-03-01:
Hi,

I have tested the latest velocity_control_sample.launch file. However, I realized that the computed velocity and real velocity do not matching.

Previously. I was using the vesc package version of [f0d983c].

(https://github.com/sbgisen/vesc/commit/f0d983c0ef787e3cb42fa917bcbfd7a7651f62a0)
[on Dec 14, 2022](https://github.com/sbgisen/vesc/commit/f0d983c0ef787e3cb42fa917bcbfd7a7651f62a0)]

In this version,  the computed velocity and real velocity are matching.

The following parameters are the same for both cases.

      num_rotor_poles_pairs (half for the latest version)
      gear_ratio
      torque_const

May I know what is the problem?

Thanks.

Tacha-S, 2023-03-02:
Please show me the result of `rosparam get /vesc_hw_interface_node` in both versions.
Also do you know what the measured and actual speed values are? I would appreciate any relationship such as the measured value is greater than the actual speed.

David9696, 2023-03-02:
I would update the rosparam once I have the access to my setup.

As a quick reply, this is what I have found:

1. The odometry/position estimation is correct.
2. The actual speed is about 2X lesser than the command speed.

Thanks.

jsupratman13, 2023-03-02:
Sorry, forgot to fix the conversion bug on vesc_hw_interface.cpp
https://github.com/sbgisen/vesc/blob/fb520b663c0c5df1a03cc46ccebb9d08fce110c7/vesc_hw_interface/src/vesc_hw_interface.cpp#L307

should be
```cpp
const double velocity_rpm = values->getVelocityERPM() / static_cast<double>(num_rotor_poles_ / 2);
```

jsupratman13, 2023-03-03:
@David9696 

I opened PR #78 which should fix the velocity conversion bug. Let us know if this fixes your environment.

David9696, 2023-03-04:
Hi @jsupratman13 ,

Yes, is working correctly now. Thanks for your great help.

PR #78 (2023-03-06): fix velocity conversion

## Change Summary
- fix #77 

## Details

## Impacts

## References

## Additional Information
closed 2023-03-04 by David9696