sbgisen/vesc #85: Differential Drive Robot with Low Torque During Low Speed On the Spot Rotation

David9696, 2023-04-25:
## Abstract

I have a differential drive robot setup by using 2 vesc controllers. The robot has difficulty to rotate on the spot with low angular speed. High angular speed is fine(>0.5 rad/s)

## How to Reproduce the Bug

Control the robot with low angular speed.

## Problem due to the Bug

Robot seems to have no enough torque for low angular speed(<0.5 rad/s) on the spot rotation. Moving forward and backward looks good.

## Suggestion

Is is possible to increase the torque with this package for low speed on the spot rotation?

May I know how to tune the parameter of "torque_const" in the velocity_control_sample.launch file?

Thanks

David

jsupratman13, 2023-04-28:
I assume you're using `command_mode: velocity` aka VelocityJointInterface? If so, tuning `torque_const` won't help you; that parameter is exclusive in `command_mode: effort`. Try adjusting your gear ratio instead. If you have ample power and your motor is not moving as expected, you should change your firmware configuration via vesc tool.

David9696, 2023-04-29:
Hi @jsupratman13 

Thanks for your reply.

Yes, I am using command_mode: velocity. Seems like the torque is not enough for on the spot rotation for a differential drive robot.

Just wandering, for a differential drive robot, would it be better(more torque during on the spot rotation) if I use velocity_duty_control? My payload requirement is 100 KG with four wheels.

Secondly, may I know what is the correct way to set the gear_ratio? Is it based on given motor specification?

From the readme file, it states that:

gear_ratio (double, default: 1.0): ratio of reduction calculated by joint velocity/motor velocity

How do I get joint velocity and motor velocity? Is it from vest tool?

Thanks.

jsupratman13, 2023-05-08:
> Just wandering, for a differential drive robot, would it be better(more torque during on the spot rotation) if I use velocity_duty_control? My payload requirement is 100 KG with four wheels.

It's probably better to add (or use stronger) suspensions if your motors are unable to handle the payload. But to answer your question, if you don't care about your robot moving at a consistent speed or if your motors can guarantee consistent speed in various payloads, velocity control is enough. If that is not the case, you should probably use either effort control (interface with effort_controllers::JointVelocityController for this case) or velocity duty control though you will have to tune the PID control properly. Note that we don't have documentation on using velocity_duty_control yet, so you will have to read the code to understand its parameters.

> Secondly, may I know what is the correct way to set the gear_ratio? Is it based on given motor specification?
From the readme file, it states that:
gear_ratio (double, default: 1.0): ratio of reduction calculated by joint velocity/motor velocity
How do I get joint velocity and motor velocity? Is it from vest tool?

Sorry, we will need to update the document sometime later.
Your gear ratio will depend on your hardware, the gears from your motor shaft to your designated joint/motor velocity. If you use external gears, include them. If you use in-wheel motor, you will need to open it and check the gears if the motor spec sheet doesn't them.