sbgisen/vesc #90: ros2_control for vesc motor controller not working with two vescs

Persius48, 2023-09-05:
I am currently working on a project to make a four-wheeled differential drive robot using 4 VESCs with ROS2 as the means of communication. Two vesc will be masters and the rest two will be slaves. I used the vesc_hw_interface with humble-devel for one VESC and it was okay. but when I tried to use two VESCs with some modification in the ros2_control_xacro by including two hardware and two joints, there was an error like the following

```
Invalid checksum
[ros2_control_node-1] [ERROR] [1693755709.123338959] 
[VescHwInterface]: Out-of-sync with VESC, discarding 49 bytes.
[ros2_control_node-1] [ERROR] [1693755709.128419540] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.128438065] 
[VescHwInterface]: Invalid payload length
[ros2_control_node-1] [ERROR] [1693755709.128448735] 
[VescHwInterface]: Out-of-sync with VESC, discarding 21 bytes.
[ros2_control_node-1] [ERROR] [1693755709.163161253] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.163192823] 
[VescHwInterface]: Out-of-sync with VESC, discarding 8 bytes.
[ros2_control_node-1] [ERROR] [1693755709.203127802] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.203179020] 
[VescHwInterface]: Out-of-sync with VESC, unknown data leading 
valid frame. Discarding 1 bytes
```
I tried using the controllers (joint_state_broadcaster and velocity_controller), which worked with a single VESC, launching it with two VESCs with two ros2_control tags including two separate actuator-type hardware. I also tried to change the velocity controller to diff_drive_controller and tried the same thing with two VESCs. But it fails every time. I am attaching the screenshot of my configuration and xacro file. Any suggestion will be really helpful.

This is the config file for the controller manager and xacro file for the resource manager that works with one vesc.

```
controller_manager:
  ros__parameters:
    update_rate: 100 # Hz
    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster
    joint_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

# Publish all joint states -----------------------------------
joint_state_broadcaster:
  ros__parameters:
    publish_rate: 100

joint_velocity_controller:
  ros__parameters:
    joints:
      - left_wheel_joint
    publish_rate: 100
```

The xacro that works with one vesc
```
<?xml version="1.0" encoding="UTF-8"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">
  <ros2_control name="vesc" type="actuator">
    <hardware>
      <plugin>vesc_hw_interface/VescHwInterface</plugin>
      <param name="port">/dev/ttyACM1</param>
      <param name="gear_ratio">1</param>
      <param name="command_mode">velocity_duty</param>
      <param name="joint_type">continuous</param>
      <param name="num_hall_sensors">3</param>
      <param name="num_rotor_poles">20</param>
      <param name="motor/Kp">0.1</param>
      <param name="motor/Ki">0.01</param>
      <param name="motor/Kd">0.02</param>
      <param name="motor/i_clamp">1.0</param>
      <param name="motor/duty_limiter">1.0</param>
      <param name="motor/antiwindup">true</param>
      <param name="motor/control_rate">100.0</param>
      <param name="motor/enable_smooth_diff">true</param>
      <param name="motor/smooth_diff/max_sample_sec">0.2</param>
      <param name="motor/smooth_diff/max_smooth_step">10</param>
[... 148 lines cut ...]
      <param name="motor/control_rate">50</param>
      <param name="motor/enable_smooth_diff">true</param>
      <param name="motor/smooth_diff/max_sample_sec">0.2</param>
      <param name="motor/smooth_diff/max_smooth_step">10</param>
      
    </hardware>
    
    <joint name="right_wheel_joint">
      <command_interface name="velocity">
        <param name="min">-5</param>
        <param name="max">5</param>
      </command_interface>
      <state_interface name="position"/>
      <state_interface name="velocity"/>

    </joint>

  </ros2_control>
[ros2](https://robotics.stackexchange.com/questions/tagged/ros2)[ros-control](https://robotics.stackexchange.com/questions/tagged/ros-control)[differential-drive](https://robotics.stackexchange.com/questions/tagged/differential-drive)[ros-humble](https://robotics.stackexchange.com/questions/tagged/ros-humble)[esc](https://robotics.stackexchange.com/questions/tagged/esc)
[Share](https://robotics.stackexchange.com/q/103891/36177)
[Edit](https://robotics.stackexchange.com/posts/103891/edit)
Delete
Flag
[edit](https://robotics.stackexchange.com/posts/103891/revisions)
```

jsupratman13, 2023-09-06:
> Invalid checksum
[ros2_control_node-1] [ERROR] [1693755709.123338959] 
[VescHwInterface]: Out-of-sync with VESC, discarding 49 bytes.
[ros2_control_node-1] [ERROR] [1693755709.128419540] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.128438065] 
[VescHwInterface]: Invalid payload length
[ros2_control_node-1] [ERROR] [1693755709.128448735] 
[VescHwInterface]: Out-of-sync with VESC, discarding 21 bytes.
[ros2_control_node-1] [ERROR] [1693755709.163161253] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.163192823] 
[VescHwInterface]: Out-of-sync with VESC, discarding 8 bytes.
[ros2_control_node-1] [ERROR] [1693755709.203127802] 
[VescHwInterface]: Invalid end-of-frame character
[ros2_control_node-1] [ERROR] [1693755709.203179020] 
[VescHwInterface]: Out-of-sync with VESC, unknown data leading 
valid frame. Discarding 1 bytes

Does this error log appear once or does it stream continuously?

Tacha-S, 2023-09-06:
If VESC works when there's only one but consistently shows such a log when there are two or more, I have seen similar behavior before. I don't have time for an in-depth investigation, but first, try adding VESC to the `dialout` group in udev rules and then test it again.

Persius48, 2023-09-06:
> Does this error log appear once or does it stream continuously?

Yes, the error message is continuous.

Persius48, 2023-09-07:
> If VESC works when there's only one but consistently shows such a log when there are two or more, I have seen similar behavior before. I don't have time for an in-depth investigation, but first, try adding VESC to the `dialout` group in udev rules and then test it again.

The vescs were already in the `dialout `group in udev and I added the current user in the `dialout `group. The problemis persistent. One vesc conntects, but two does not.

erknl, 2024-03-29:
Is the problem solved?

Tacha-S, 2024-04-01:
We also tested two VESCs as a differential drive and did not observe similar issues. Regarding the advice to add the user to the dialout group, there was only a comment that it had been added, without any logs on how the udev rules were set. This is insufficient information for us to replicate the issue, so please provide more detailed information, logs, and settings that would allow us to reproduce the problem.

onmeraz, 2025-02-06:
> Мы также протестировали два VESC в качестве дифференциального привода и не наблюдали подобных проблем. Что касается совета добавить пользователя в группу dialout, был только комментарий о том, что он был добавлен, без каких-либо журналов о том, как были установлены правила udev. Этой информации недостаточно для того, чтобы мы могли воспроизвести проблему, поэтому, пожалуйста, предоставьте более подробную информацию, журналы и настройки, которые позволят нам воспроизвести проблему.

Hello, can you provide example code with differential drive please?

Tacha-S, 2025-02-14:
Which branch do you need the example from?
Also, please provide the original text for the quote.