sbgisen/vesc #98: Use position and velocity control on vesc

nyxrobotics, 2024-09-18:
## Abstract
I want to use the PID controller implemented inside the VESC for speed and position control.
I remember that in the past there were some parts that changed depending on the version of the firmware, so you need to check carefully.
I probably needed to copy vesc_packet.cpp and vesc_packet.h from the latest bldc firmware.
I would like to add this to the README as I vaguely remember it.

## Purpose

## Implementation Details

nyxrobotics, 2024-09-18:
https://github.com/sbgisen/vesc/issues/43#issuecomment-1271434529

nyxrobotics, 2024-09-18:
firmware: [datatypes.h](https://github.com/vedderb/bldc/blob/master/datatypes.h)
vesc_driver: [data_map.h](https://github.com/sbgisen/vesc/blob/develop/vesc_driver/include/vesc_driver/data_map.h)

It would be good to have a script that automatically generates this header

nyxrobotics, 2024-09-18:
Since sending position commands has been implemented, it would be usable if the home position adjustment function is implemented. I think the origin alignment function needs to send a command to vesc to set the current position.
https://github.com/sbgisen/vesc/blob/48bf2853b32254e7214fdb4d0ce1824af00e4f06/vesc_driver/src/vesc_driver.cpp#L243

jsupratman13, 2025-02-13:
Any progress in this?

nyxrobotics, 2025-02-13:
I am going to have @saitenntaisei do it.
However, it may be difficult to do so now, as the class design may need to be changed in the #106 PR.

jsupratman13, 2025-02-20:
In the original code, it seems that you cannot use (PID) position control, if you don't have an encoder. I believe this means that the position control will not work for hall sensors either.

It addition, it seems that the position range is limited to 0 ~ 360deg. 

regarding encoder and position control
https://github.com/vedderb/bldc/blob/022bd134ba1233c9375933491782b7e39fd4fd33/motor/mcpwm_foc.c#L763-L771
https://www.vesc-project.com/node/258
https://vesc-project.com/node/4171

regarding range
https://github.com/vedderb/bldc/issues/319

jsupratman13, 2025-02-20:
> In the original code, it seems that you cannot use (PID) position control, if you don't have an encoder. I believe this means that the position control will not work for hall sensors either.

I retract this statement, it seems you can use position control with hall sensors too.
Need to properly set the following settings on vesc tool:
- Motor settings/Additional Info/Setup/Motor Poles
- Motor settings/Additional Info/Setup/Gear Ratio
- Motor settings/Additional Info/Motor General/Position Sensor -> set to hall sensor
- Motor settings/PID Controllers/Position Angle Division
- Motor settings/PID Controllers (the gains in general)

jsupratman13, 2025-03-06:
section where position angle division is used to update current position angle
https://github.com/vedderb/bldc/blob/b214b8d91e39ab5d227d28cd40f85c3d60b3d1a8/motor/mcpwm_foc.c#L3658-L3674

PR #115 (2025-04-09): use vesc's pid position/velocity control

## Change Summary

- Introduce/fix the native "position"/"velocity" command mode where hardware interface uses vesc's position/velocity pid control instead of using the pid control in vesc_wheel_controller/vesc_servo_controller. 
- Note the new feature assumes that the vesc firmware version is 5.03.
- Resolves #98 

## Details
- [vesc_driver] added interface to extract MC configuration from vesc firmware
- [vesc_driver] added function to get vesc PID position
- [vesc_hw_interface] added new "position" command mode, which relies on vesc pid pos control  [... cut]
closed 2025-04-09 by Tacha-S