sbgisen/vesc #105: question the package is still supported

sipmine, 2024-12-28:
Today I tried to build it under ros2 jazzy under docker container, and it was a pain, 1 function did not have void* return, so colcon build gave warning and did not build the package, also none of the launch files could be built.

Tacha-S, 2025-01-08:
On the branch [jazzy-devel](https://github.com/sbgisen/vesc/tree/jazzy-devel)?

sipmine, 2025-01-08:
some of the lunch files are not translated to support ros2

Tacha-S, 2025-01-08:
Sure, only `.launch.py` files are currently supported in ROS2.
It is expected that support will gradually be expanded in the future, but the timeline is uncertain.

If you could add support and submit a PR, it would be greatly appreciated.

sipmine, 2025-01-08:
Yeah, well, I'll get to it soon.