sbgisen/vesc #114: Discrepancy between velocity calculation methods in wheel controller

Tacha-S, 2025-03-27:
## Abstract

The `getVelocityERPM` function is used to retrieve the motor's velocity in electrical RPM (ERPM) from the VESC packet. This value is then converted into mechanical velocity (rad/s) using the number of rotor pole pairs and the gear ratio. However, in the PID control logic, velocity is estimated by differentiating `position_steps_`, which represents the encoder's step count.

These two velocity values ([velocity_sens_](https://github.com/sbgisen/vesc/blob/1c4bf5dd174956bf46da15023e5f6de0961372c5/vesc_hw_interface/src/vesc_wheel_controller.cpp#L256) derived from `getVelocityERPM` and the [velocity](https://github.com/sbgisen/vesc/blob/1c4bf5dd174956bf46da15023e5f6de0961372c5/vesc_hw_interface/src/vesc_wheel_controller.cpp#L151) calculated from `position_steps_`) may not match due to differences in their data sources and calculation methods. The velocity from `getVelocityERPM` is directly obtained from the motor's internal sensors, while the velocity from `position_steps_` depends on the encoder's resolution and the sampling rate.

This discrepancy raises questions about the consistency and reliability of the velocity values used in the control logic. It may be necessary to investigate whether these differences impact the performance of the PID controller and determine which velocity source is more accurate or suitable for the application.

## Purpose

## Implementation Details

jsupratman13, 2025-03-27:
VescStepDifference smoothing filter also impact the velocity calculation (calculated from position_steps).

Tacha-S, 2025-03-27:
I understand what each of them is doing.  
If the smoothing is effectively removing disturbances better than the ERPM reports from the motor, then wouldn't it make more sense to report the smoothed `current_vel` to the joint states instead of `velocity_sens_`?  
What I'm questioning is not the meaning of each current velocity data, but whether it really makes sense for the hardware interface to have an inconsistent understanding of the current velocity.

jsupratman13, 2025-03-27:
I don't think the smoothing itself was design to be better than the erpm reported by the motor, it was designed to work with the tachometer noise.

That being said you're right to question this.

If my memory serves correctly, back then (i believe it was when the previous author work on vesc firmware 3?), the only reliable data from vesc was the tachometer value, hence why this package relies so much on the tachometer.

@nyxrobotics is this correct? I believe you were the one working on this back then.

jsupratman13, 2025-03-27:
There is a discussion about wanting to use the vesc pid speed control here #98

nyxrobotics, 2025-04-07:
velocity_sens_ is a function that uses coil voltages and other factors to estimate the motor's turning speed.
The values are linear, but there is a problem that integration does not match the angle.
Therefore, in PID control, the integrals do not converge and feedback is not correct.
Since my program was intended for PID control, the value of the Hall sensor counter (variable called tachometer in vesc) is divided by time to get the speed.